adrianmo/go-nmea #8: Doesn't seem to handle null fields.

icholy, 2017-06-07:
I have the following sentence:

`$GPRMC,142754.0,A,4302.539570,N,07920.379823,W,0.0,,070617,0.0,E,A*3F`

Which fails to parse with: `GPRMC decode course error:` due to the parser not handling null fields correctly. The easiest solution would be to use `0` when there's a null field, but idk if that's a good idea.

icholy, 2017-06-14:
@adrianmo the [gpgsv](https://github.com/adrianmo/go-nmea/blob/adc20ca878e6cd52866b3c540ba8dcd077ad550a/gpgsv.go#L47-L52) parsing code checks for empty strings before parsing. Would you accept a PR adding these checks to all fields?

adrianmo, 2017-06-15:
Good catch @icholy! 

I would definitely accept the PR with the proposed solution. Even though that will set a [zero value](https://tour.golang.org/basics/12) to those fields. 

In the future we should probably change the struct fields to pointers instead of values. That empty fields will be <nil>.

@icholy do all values in a NMEA sentence accept an empty string? Can you fix the rest of sentence types that have the same issue? Can you add a couple of tests with empty values sentences?

Thanks!

icholy, 2017-06-15:
@adrianmo I'm pretty sure not all fields are allowed to be null, but I'm having a hard time finding explicit documentation.

adrianmo, 2017-06-15:
@icholy OK, let's at least check for an empty string and prevent the lib from crashing.

icholy, 2017-06-15:
@adrianmo I'm looked at the parser in https://github.com/cloudyourcar/minmea, and it just assigns zero values if the fields are null.

adrianmo, 2017-06-15:
@icholy great - let's do the same then.

PR #9 (2017-06-19): Allow null values in GPRMC speed & course fields
Fixes #8
closed 2017-06-19 by adrianmo