adrianmo/go-nmea #38: Make GPGLL struct fields JSON-serializable by adding corresponding tags

denismakogon, 2018-07-17:


icholy, 2018-07-18:
`encoding/json` works without the tags.

adrianmo, 2018-07-18:
Hi @denismakogon, 

As @icholy said, you can already encode a sentence struct, GPGLL in your case, to a JSON string. Here's a code snippet that shows how to do it (ignoring errors).

```golang
sentence := "$GPGLL,3926.7952,N,12000.5947,W,022732,A,A*58"
s, _ := nmea.Parse(sentence)
m := s.(nmea.GPGLL)
b, _ := json.Marshal(m)
fmt.Printf("%s\n", string(b))
```
Output:
```json
{"Type":"GPGLL","Fields":["3926.7952","N","12000.5947","W","022732","A","A"],"Checksum":"58","Raw":"$GPGLL,3926.7952,N,12000.5947,W,022732,A,A*58","Latitude":39.44658666666667,"Longitude":-120.00991166666667,"Time":{"Valid":true,"Hour":2,"Minute":27,"Second":32,"Millisecond":0},"Validity":"A"}
```

Let us know if this is what you were looking for.

Cheers!