adrianmo/go-nmea #39: Latitude and Longitude in Decimal Format

adityaalviori, 2018-07-20:
Hi @adrianmo , 
May I get the GPS output on decimal format?

Thanks for your help

Regards

icholy, 2018-07-23:
Can you post an example of the desired output?

adityaalviori, 2018-07-23:
Hi @icholy, 
I want the output like this: -6.21462, 106.84513
Thanks

adrianmo, 2018-07-23:
Hi @adityaalviori,

By default, the library parses the lat/long coordinates to float, so you can just use the `Latitude` and `Longitude` fields of the parsed sentence. See the example below:

```
sentence := "$GPRMC,220516,A,5133.82,N,00042.24,W,173.8,231.8,130694,004.2,W*70"
s, _ := nmea.Parse(sentence)
m := s.(nmea.GPRMC)
fmt.Printf("Latitude decimal: %f\n", m.Latitude)
fmt.Printf("Longitude decimal: %f\n", m.Longitude)
```

Output:
```
Latitude decimal: 51.563667
Longitude decimal: -0.704000
```
Let us know if this is what you are looking for.
Thanks.

adityaalviori, 2018-07-23:
Hi @adrianmo 

Yeah, i want the output like this format. 
Thank you very much for your help

Regards

adityaalviori

closed 2018-07-23 by adityaalviori