adrianmo/go-nmea #43: Accept other talkers than "GP"

sthorshaug, 2018-12-18:
The NMEA 0183 protocol offers multiple talkers based on which equipment generated the message.  
_GP_ indicates a GPS receiver, _GL_ a GLONASS receiver, _GN_ a combined GNSS receiver, _IN_ an integrated navigation receiver and so fort. See [here](http://freenmea.net/docs) for examples of different talkers.

As of today, go-nmea only accepts GP as talker. Is there an easy way of allowing different talkers?  

An easy (but dirty) way is of course to manipulate the talker into GP.

icholy, 2018-12-18:
@sthorshaug as it stands, there are separate message types for each prefix (even if the structure is identical). For example [GNRMC](https://github.com/adrianmo/go-nmea/blob/master/gnrmc.go) and [GPRMC](https://github.com/adrianmo/go-nmea/blob/master/gprmc.go). This is something we should fix.

sthorshaug, 2018-12-19:
I'm currently learning Go, but I would be happy to help if help is needed. I'm a C-programmer, so the framework and syntax in Go is a bit different from what I'm used to :-).  

I'm planning to add support for a few new messages, but I'll wait with adding them until the talker-issue have been resolved to avoid more work at a later point.

icholy, 2018-12-19:
@sthorshaug so are the first two characters always going to be the talker? Or are there message formats that don't conform?

adrianmo, 2018-12-19:
@sthorshaug thanks for bringing this up, I was not aware of that when I started this library. It definitely doesn't make much sense to have multiple implementations of the same message due to the talker ID, as we have today with [GNRMC](https://github.com/adrianmo/go-nmea/blob/master/gnrmc.go) and [GPRMC](https://github.com/adrianmo/go-nmea/blob/master/gprmc.go, for example, as @icholy mentioned.

However, I have the same concern as @icholy: are the first two characters always going to be the talker? do different talkers implement the messages equally or there are known differences?

I'll also do some investigation, but I'm not an expert in the subject and would like to hear some real experiences...

sthorshaug, 2018-12-19:
@icholy @adrianmo 
At my work I write code that both encodes and decodes NMEA sentences.  
NMEA 0183 is a combined electrical and data specification for communication between marine electronic devices. It has different versions, and from one version to another, some mesasges may have received extra fields (but existing fields are not removed nor have their order changed). The specification is widely adpoted for other equipment using GNSS-receivers.

In NMEA 0183 the talker is always two characters, except when the first character is _P_. The character _P_ indicates a properitary messaage.

The content of the messages should not change based on the talker, since the messages are defined by the specification. The purpose of the talker, is to identify the sender of the message.  
There may of course be that some vendors have either misinterpreted the specification, or by choice selected to use the fields inside a message differently (for such as null fields).

icholy, 2018-12-19:
@sthorshaug So in the case of proprietary messages like [$PGRME, $PGRMM, $PSLIB, etc ...](http://aprs.gids.nl/nmea/#rme) we can use "P" as the talker ID and the rest as the message data type right?

sthorshaug, 2018-12-19:
@icholy As far as i know, that is correct. Since we are talking about proprietary messages here, there are of course a lot of variations. I don't know all proprietary messages, but I know that some data types have 4 letters (as in $P**GRME**) and some have 3 letters (as in $P**SXN**).  
And to make things even worse, some data types can not be identified by their talker and data type, you need eg a number as well (such as _$PSXN,10_ and _$PSXN,23_).
And this is why are they are called proprietary messages :-).

icholy, 2018-12-21:
@adrianmo @sthorshaug here's what I'm thinking. We generalize the message parsing so that we can reuse message types while making sure all the tests pass. After than we add support for more talkers in separate PR's as people want/need them.

sthorshaug, 2018-12-22:
@icholy Generalization of the message parsing is the first step. 
I have an idea how the rest could be solved swirling in the back of my head, but I have to test it first. We could perhaps accomplish backward compability and allow new talkers without specific PRs.
I’m without my computer for a few days, but I’ll test my idea a few days after Christmas.

sthorshaug, 2018-12-27:
My idea for backward compability wasn't working.  
I think you to have to choose between backward compability, using PRs or perhaps introducing new enums (e.g. one enum for talker, one enum for base message type (GGA, GSA etc) and one enum combing those into todays type).  

Of course the last option is to do nothing, and keep the library as it is.

icholy, 2018-12-30:
@sthorshaug if we exported possible talkers, we could defs create a new types for them. something like:

``` go
type Talker string

const TalkerGP = Talker("GP")
// etc ...
```

What's the use case thought?

sthorshaug, 2019-01-02:
@icholy I want to accomplish the following:

- Decode NMEA sentences with "any" talker (only accepted talkers is fair enough)
- Identify the talker of the decoded sentences, since the talker reveal important information 
- Don't overcomplicate things, and preferably avoiding PRs when adding talkers

icholy, 2019-01-05:
mission accomplished 👍

closed 2019-01-05 by icholy