adrianmo/go-nmea #45: Replace testify with gotestools

icholy, 2019-01-05:
I want to replace [testify](https://github.com/stretchr/testify) with [gotesttools](https://github.com/gotestyourself/gotest.tools). It uses [google/go-cmp](https://github.com/google/go-cmp) instead of [reflect.DeepEqual](https://golang.org/pkg/reflect/#DeepEqual) which lets you override the comparison function for specific types. This will allow us to have a single test helper for comparing message types instead of rolling separate ones. It also outputs nicer diffs.

adrianmo, 2019-01-07:
My only concern is about the maturity and adoption of the project. Testify has been around for a long time and has a very wide adoption, 6400+ stars vs 70+ stars. Do you have examples of other projects migrating from Testify to gotest.tools? Or just well-known projects using gotest.tools?

icholy, 2019-01-07:
Docker uses it.

adrianmo, 2019-01-08:
Fair enough.

closed 2020-05-11 by icholy