adrianmo/go-nmea #56: sentence checksum error reported, though both values seem the same

madwizard, 2019-06-17:
I am running your example code on GNRMC data received from my GNSS module.
On nmea.Parse(sentence) I get:

```
]019/06/17 21:26:23 Couldn't parse GPS data: nmea: sentence checksum mismatch [79 != 79
2019/06/17 21:26:24 $GNRMC,192624.000,A,5327.813301,N,01432.630601,E,0.00,325.41,170619,,,A*7E
]019/06/17 21:26:24 Couldn't parse GPS data: nmea: sentence checksum mismatch [7E != 7E
2019/06/17 21:26:25 $GNRMC,192625.000,A,5327.813301,N,01432.630601,E,0.00,325.41,170619,,,A*7F
]019/06/17 21:26:25 Couldn't parse GPS data: nmea: sentence checksum mismatch [7F != 7F
^C2019/06/17 21:26:25 main: Got signal: interrupt
2019/06/17 21:26:26 $GNRMC,192626.000,A,5327.813301,N,01432.630601,E,0.00,325.41,170619,,,A*7C
]019/06/17 21:26:26 Couldn't parse GPS data: nmea: sentence checksum mismatch [7C != 7C
^C2019/06/17 21:26:27 $GNRMC,192627.000,A,5327.813301,N,01432.630601,E,0.00,325.41,170619,,,A*7D
]019/06/17 21:26:27 Couldn't parse GPS data: nmea: sentence checksum mismatch [7D != 7D
```

Both checksums seem the same to me in the output above. 
The place I pasted your example code: 

https://github.com/madwizard/zut_wifi_logger/blob/2aead364882076d5a836c105a30cb6760d14c266/gps.go#L42

madwizard, 2019-06-17:
Adding TrimSuffix for trimming new line in this line:
https://github.com/adrianmo/go-nmea/blob/9861e243767ddcbc01ec662adc18b97a35f2bbe7/sentence.go#L70

Fixed it for me:

`        checksumRaw = strings.TrimSuffix(strings.ToUpper(raw[sumSepIndex+1:]), "\n")
`

icholy, 2019-06-17:
Does the string you're parsing have a tailing newline? You should be using a `bufio.Scanner`.

madwizard, 2019-06-17:
Ok, thank you, I'll check it.

closed 2019-06-17 by icholy