adrianmo/go-nmea #73: Support custom message types

yavosh, 2020-05-01:
I would like to extend this lib to support custom message parsing, is this something you would consider merging. We are using this to parse some custom message which are not part of the NMEA standard. The idea is to have a way to register parsers for new message types.

adrianmo, 2020-05-04:
Hi @yavosh, thanks for the interest in the library. What kind of messages are you considering implementing? Is this something like the PMTK (Mediatek) or PGRME (Garmin) proprietary messages? If that's the case, yes, we will review your contribution and eventually merge it.

I also like the idea of having the possibility to extend the library via external plugins or modules, but that would require a more in-depth design session.

yavosh, 2020-05-04:
No these are proprietary messages we generate ourselves, they don't really fit NMEA in any way otherwise we would be contributing them.

yavosh, 2020-05-04:
The extensibility suggestion is to replace implementation of `func Parse(raw string) (Sentence, error)` from the current switch on type to a `map[string]func()` where each existing supported message type would have its constructor added and there can be a method like `RegisterParser` which would allow you to add a new message type with its parser.

Should be fairly unobtrusive change.

adrianmo, 2020-05-05:
Your suggestion looks good. If you are willing to put together a draft/prototype we will definitely review it and help with the integration. If not, no worries, we will add it to our backlog.

icholy, 2020-05-05:
Being able to register parsers is something I've wanted for a while.

yavosh, 2020-05-05:
Hey @adrianmo have a look at how it could work https://github.com/adrianmo/go-nmea/pull/75

closed 2020-05-11 by icholy