adrianmo/go-nmea #83: DPT third field depends on nmea version?

aldas, 2021-12-08:
DPT documentation refers to https://gpsd.gitlab.io/gpsd/NMEA.html#_dpt_depth_of_water and in that is example `$INDPT,2.3,0.0*46`

that example is not parseable by the library. I get `nmea: INDPT invalid range scale: index out of range`

Some other documents ala https://www.tronico.fi/OH6NT/docs/NMEA0183.pdf
describe DPT as
[image]

It would be nice if DTP parsing can handle 2 and 3 fields gracefully without erroring.

icholy, 2021-12-08:
@aldas what sensor is outputting those sentences?

aldas, 2021-12-08:
Well, I am writing diagnostics software to read some echosounder values. At first we intent to use "Skipper FDS-101 Echosounder" which uses 3 fields in DPT but when writing software I saw this inconsistency when writing unit-tests against spec + manual examples. Vessels that we deal with have wide range of electronics and god knows when it would 3 fields and when 2 fields.

From googling it seems that these 2 field ones do exist https://forum.arduino.cc/t/nmea-depth-sensor/267009 `$SDDPT,004.8,*75`

icholy, 2021-12-08:
Thanks for the details, we'll get this fixed for you. If you're short on time, you can unblock yourself by registering a custom parser for the DPT sentences https://github.com/adrianmo/go-nmea#custom-message-parsing

``` go
err := nmea.RegisterParser("DPT", func(s nmea.BaseSentence) (nmea.Sentence, error) {
    p := NewParser(s)
    p.AssertType(TypeDPT)
    dpt := DPT{
	BaseSentence: s,
	Depth:        p.Float64(0, "depth"),
	Offset:       p.Float64(1, "offset"),
    }
    if len(p.Fields) > 2 {
	dpt.RangeScale = p.Float64(2, "range scale")
    }
    return dpt, p.Err()
})
if err != nil {
    panic(err)
}
```

PR #84 (2021-12-10): Allow omitting the DPT scale field
Fixes #83
closed 2021-12-10 by adrianmo