adrianmo/go-nmea #89: parsing errors

aldas, 2022-01-02:
About errors. When parsing fails currently only first error is stored and all subsequent errors are discarded.  I propose if `p.EnumString` and other parsing method errors would be distinguishable from each other and parsers would store all errors.  it would allow you to introspect what went wrong and some circumstances ignore some error types and still use the parsed sentence.

maybe just wrapping them (or storing into slice or multierror)
```go
func (p *Parser) SetErr(context, value string) {
	if p.err == nil {
		p.err = fmt.Errorf("nmea: %s invalid %s: %s", p.Prefix(), context, value)
	} else {
		p.err = fmt.Errorf("nmea: %s invalid %s: %s, err: %w", p.Prefix(), context, value, p.err)
	}
}
```

icholy, 2022-01-04:
I don't like the error wrapping approach because it will break all the existing tests and the error messages will be very ugly. The slice/multi-error approach will let use keep backwards compatibility. However, I'd like to understand the "why" a bit more. Can you describe your use case?

aldas, 2022-01-04:
when parsing fails it stops at first problem. which means that if there are more than 1 field problematic I would need to parse sentence multiple times to know what all problems are. In production environment this is impossible or tedious - especially if you are doing first assessment of the problem scope.

-----
in certain situation when you use library just as a simple converter-gateway to another system (ala serial reader -> nmea parser -> mqtt -> ...) you want to delegate decision making from library to another component upstream. It is not possible to make decision if sentence that was parsed has incomplete values because parsing actually stopped after first error and relevant information for you is actually ok.  For example - some enum fields can contain values that go-nmea does not recognize. I know from reading device manuals that sometimes manufacturers only document enum values that their device uses and left out values that are specified by NMEA. Sometimes values change - ala new GNSS constellations are launched. I think 15 year ago there were 4 constellations - now there are 6. Although gnss constellations are not enums, but there are cases where constellation is certain place at value for example `GNS.Mode` field. Let alone obscure devices that send their custom values. I think there are already reported issues that there are devices that do not send CRC. Everything is easy when software is meant to use specific hardware - but situation where devices vary it gets harder as you can not predict what devices you are using in future.

Currently parsing methods return early when parser has already registered an error 
```go
func (p *Parser) Int64(i int, context string) int64 {
	s := p.String(i, context)
	if p.err != nil {
		return 0
	}
```
but maybe this could be made "optional" by some flag or default not to stop.

icholy, 2022-01-04:
Ok, that makes sense to me. If you want to put together a PR I'll review.

closed 2022-03-12 by aldas