adrianmo/go-nmea #98: processing multiline / segmented messages

metemaad, 2022-08-11:
I could not find an example for processing AIS 5 message here. This type of message has more than one line for example:

\g:1-2-3730,s:43576,c:1654340281,t:1654340381*19\!AIVDM,2,1,7,,569EH`8000009aGUB20IF1UD4r1UF3OK7>22220N4PT38t0000000000,0*42

\g:2-2-3730*5A\!AIVDM,2,2,7,,000000000000000,2*62

Is there any way to process such cases?

nilsmagnus, 2023-08-17:
Same here, I am not able to parse AIS 5 messages of this kind

nilsmagnus, 2023-08-17:
Since this issue is solved in the master branch, could you add a new tag?

icholy, 2023-08-17:
cc @adrianmo

kimgr, 2024-03-16:
> Since this issue is solved in the master branch, could you add a new tag?

@nilsmagnus Could you point me to where this is solved? I haven't found any facilities in go-nmea for multiline reassembly, and I'm curious if I've just been looking in the wrong places. Thanks!

aldas, 2024-03-16:
Try this

```go
func TestName(t *testing.T) {
	var tagBlocks []TagBlock
	p := SentenceParser{
		OnTagBlock: func(tb TagBlock) error {
			tagBlocks = append(tagBlocks, tb)
			return nil
		},
	}
	_, err := p.Parse("\\g:1-2-73874,n:157036,s:r003669945,c:1241544035*4A\\!AIVDM,1,1,,B,15N4cJ`005Jrek0H@9n`DW5608EP,0*13")
	if err != nil {
		t.Fatal(err)
	}
	fmt.Printf("%+v\n", tagBlocks)
}
```

This should output that single parsed tag block as
```
[{Time:1241544035 RelativeTime:0 Destination: Grouping:1-2-73874 LineCount:157036 Source:r003669945 Text:}]
```

kimgr, 2024-03-16:
@aldas Thanks --

> assembly is up to you to implement.

Ah, yeah, then I hadn't missed anything. I got the impression from above that there was reassembly support built in somewhere. Thanks for clarifying.