adrianmo/go-nmea #109: Fails to parse IIXDR

Piorosen, 2023-07-17:
occur error when parsing data.
i received error message, "nmea: IIXDR invalid transducer type:  A"

my gps data sender is providing bellow data
$IIXDR, A, 0.0, D, ROLL, A, 0.0, D, PITCH*15

aldas, 2023-07-17:
every field has really leading space?

Piorosen, 2023-07-17:
yes, because nmea data is provided by simulator.

aldas, 2023-07-17:
At the moment library does not try to actively trim fields from whitespaces as leading space is not something "normal". For example leading zero(s) `0` is valid thing for numbers by NMEA0183 spec but space?

`$IIXDR, A, 0.0, D, ROLL, A, 0.0, D, PITCH*15` has 8 extra spaces. Considering 9600 baudrate being common this is quite huge overhead. Each extra character at these low transfer speeds is very expensive for throughput.

I guess we could ignore leading spaces for some of the field types - number, enums but for `string` fields this leads to question - could be that leading space is intended or is it not?

Piorosen, 2023-07-17:
Thank you for letting us know what caused this.
i was able to determine the cause of the issue and resolve it. Thank you.
@aldas 

this issue is closed that solve.

closed 2023-07-17 by Piorosen